astro-pi/python-sense-hat #124: WARNING:root:Failed to initialise TCS34725 colour sensor. (sensor not present)

DMHas, 2022-08-26:
Hello together,

when I use `sense = SenseHat()` at the python program, I get this error message:

`WARNING:root:Failed to initialise TCS34725 colour sensor. (sensor not present)`

The wiring is fine. I read something about #[here ](https://forums.raspberrypi.com/viewtopic.php?t=338641). But my SenseHat is over 2 years old and it don't should have a colour sensor. Make I a mistake or is this behaviour a mistake of the SenseHat library?

Best Regards,
DMHas

G3zz, 2022-08-26:
Hi DMHas, thanks for getting in touch.

This is just a warning and not an error. You can continue to use the library as you have been doing, or if you want to ignore the warning you can change the logger level as has been suggested here: https://github.com/astro-pi/python-sense-hat/issues/121#issuecomment-1179164651.

I'll close this issue now.

closed 2022-08-26 by G3zz

DMHas, 2022-08-26:
Thanks G3zz for your answer.