astro-pi/python-sense-hat #125: sense-hat V 1.0 with latest driver

hb9cvh, 2022-10-12:
My sense-hat is now on a new Rpi 3+ with PiOS Bullseye and latest python-sense-hat.
This little program :

> from sense_hat import SenseHat
> sense = SenseHat()
> sense.show_message("Hello World")

Display the message but show: 

> WARNING:root:Failed to initialise TCS34725 colour sensor. (sensor not present)

What can I do?

bsimmo, 2022-10-13:
You ignore it.

I wish they didn't display this message as it brings a lot of worry and
comments like yours to users.

Basically they made a new version of the sensehat a year or so ago for the
ISS/AstroPi missions that has a light sensor on it, the retail (original
version) does not have it. And that is what it is saying.

They need to write in a SenseHAT version check so if it is not the v2
(which we cannot buy but is on the ISS) then do not display to user.

bsimmo, 2022-10-13:
> Hi Ben,
>
> Thanks for your answer.
>
> Unfortunately my program abort when I insert the driver sense-hat. So I
> downloaded the source from GitHub and found the lines 96 - 101 in
> sense_hat.py :
>
> # initialise the TCS34725 colour sensor (if possible)
> try:
> self._colour = ColourSensor()
> except Exception as e:
> logging.warning(e)
> pass
>
> I commended this part out and my program runs now.
>
> Sincerely
> John
>
> Johann Maurer, Hauptstrasse 85,
> CH 2563 Ipsach, Schweiz

They should change (hopefully to stop users being worried about something
they don't even haven)
logging.warning(e)
to whatever it is that doesn't normally show this error in 'logging' , then
users can override the message logging level IF they are having problems
with it and it comes to debugging the hardware.

Also Add logging/debug information in the documents.

Or have it check the SenseHAT version first and then only check for it?

:-)

But you've made me crack out my sensehats again and have a play :-)

G3zz, 2022-11-21:
Thanks for your feedback - the latest version (2.5.1) has downgraded that output to debug level, so as to not cause confusion. The documentation is in the process of being updated too

closed 2022-11-21 by G3zz