astro-pi/python-sense-hat #126: Colour/Light intensity sensor issue.

kal9001, 2022-11-18:
The Raspberry Pi Documentation for the V2 sense hat shows they use a different sensor to the Astro Pi. Instead of the TCS34725 it uses the TCS3400 which sits at a different i2c address and with different registers for it's data.

I've seen the statement that sense hats don't have the light sensor and this is why there is a warning all the time. But I can physically see it, it's on the silk screen, and the RPi docs show it. So dug deeper and was able to confirm the different part number. It's possible that others do physically have this sensor but it won't work with the drivers as they are.

I've been able to fix this on my own machine by editing the sense hat library with the correct registers, gain options, and sensor ID, all changes are wholly within the colour.py file and it now works fine as best as I can tell.

I don't know enough about github to be able to make a code change suggestion. Also my change does not allow models with the older chip to work at all. It seems more sensible to do some kind of check to scan for the old device on i2c addr 29, or the new one at 39, but the added annoyance is that the TSC3400 does not initialise itself with a name or driver in /sys/bus/i2c/ but does appear on an i2cdetect.

For anyone interested the file with the new parameters is below.
[colour.py.txt](https://github.com/astro-pi/python-sense-hat/files/10036955/colour.py.txt)

bsimmo, 2022-11-18:
The SenseHAT V1 does not have it, although the V2 (with the light sensor)
was release quite a while ago for AstroPi (I guess to groups and on the
ISS?) it has only just become available for the public to buy about a week
ago. I guess the change of sensor is supply? (Pihut had them)
Hence nobody will have seen a problem.

That's my assumption anyway.
Hence everyone else has been talking about the V1 sensehat until now.

G3zz, 2022-11-18:
Thanks for creating the issue and sharing your fix.
To summarise the current situation:

- V1 Sense Hat uses TCS34725
- V2 Sense Hat (used in the Astro Pi challenge for several years) uses TCS34725. 

Because of supply issues for the TCS34725, the "V2" sense hat available to buy uses the TSC3400. Arguably this is a different version entirely, but that's the situation!

As you suggest, the next steps are updating this library to detect which version of the light sensor is present 👍

bsimmo, 2022-11-18:
The V1 doesn't have the sensor.
Ben

kal9001, 2022-11-18:
@bsimmo 
> available for the public to buy about a week ago. I guess the change of sensor is supply? (Pihut had them)

I got mine from PiHut and i apparently the only place that offers the V2 only, and don't have the V1 listed at all. It's a shame they didn't call it the V3 perhaps to avoid confusion with the Astro Pi version. Me and a friend figured the intent was to keep the same sensor and so the same name.

@G3zz 
> the next steps are updating this library to detect which version of the light sensor is present

I'm not sure how to do that, one easy thought is to simply change the SenseHat constructor to take an argument for which part it has, but it's more useful if the user doesn't need to know/care what chip it is.
Checking /sys/bus/i2c/1- folders with the device information, then the SenseHat driver could branch on if device 1-0029 or 1-0039 is present. However we found that the TCS3400 doesn't populate anything in this folder despite showing up on the bus. Is this some config that is missing from the i2c driver in linux/raspbian itself? If so, how and where to add this. I've tried a few searches but information seems thin on the ground.

G3zz, 2022-11-23:
Hi @kal9001 , I agree the user should not need to know the details of the chips in their hat - that would be too confusing. 
I am in contact with RPL and they will update the i2c driver.

tvoverbeek, 2022-11-26:
See my PR #131 which solves this issue.