astro-pi/python-sense-hat #136: Error: Weird behavior in the sensehat

PetriJF, 2023-02-15:
Hello everyone

I am using a sense hat for one of my projects and I am getting some weird behavior. Up until yesterday everything was fine, but today the LED matrix flashed a quick white blink then changes to the rainbow until the raspberry pi boots up into ubuntu 22,04 completely. Afterwards, it just turns off. 

I thought it was weird and tried running some code. I received this error:
`OSError: IMU Init Failed`

I did an i2cdetect and this is what I get (Note that on 41 and 42 I have 2 other devices)
`     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- 41 42 -- -- -- UU -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- 5f
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- --
70: 70 -- -- -- -- -- -- --`

I checked `ls -l /dev/i2c-1` and it gives:
`crw-rw---- 1 root dialout 89, 1 Sep  9 21:49 /dev/i2c-1`

I tried `sudo nano /boot/config.txt` and added `dtoverlay=rpi-sense` to it.

Would anyone happen to know what could be causing the problem?

Thank you for your time!

bsimmo, 2023-02-16:
I don't know yet, but for diagnostics of the HAT itself Grab the latest RPi
OS and try it with that.
Which SenseHAT version 1 or 2?

bsimmo, 2023-02-16:
Try support in the RPi support forum
https://forums.raspberrypi.com/viewforum.php?f=104

PetriJF, 2023-02-16:
Hello!

Thank you for your response! I am using the raspberry pi 4b with the version 1.0 sensehat from 2015. I will try the forum as well.

PetriJF, 2023-02-16:
Regarding voltages, they are slightly higher at 3.8 and 5.9.
Would that cause the problem?