astro-pi/python-sense-hat #138: SenseHat LED matrix not responding

DMHas, 2023-03-18:
Hello all,

I have the problem that on both of my Raspberry Pi (2 &4) the Sense Hat LED matrix is no longer responsive. After powering on the Pi, the LED matrix lights up in rainbow colors. But then the LED matrix does not respond to any commands - that is, it does not display anything when you want to display text on the LED matrix. No error message is displayed.
At first I thought it was a hardware defect. For testing I connected the Sense Hat to my old Raspberry Pi 2 - it worked as usual. Then I updated the Raspberry Pi 2 (sudo apt update && suso apt upgrade). After that, the error also occurred on the Raspberry Pi 2 - as described above. It must be a software / driver problem. The error occurs with a Sense Hat V1 like Sense Hat V2.

Have any of you experienced the same phenomenon or have an idea how I can narrow down or fix the error?
(Temperature, humidity and pressure sensor are working normally).

Greetings DMHas

Translated with www.DeepL.com/Translator (free version)

aldenmoreton, 2023-03-22:
I have experienced the same problem on a Raspberry Pi 4 with a Sense Hat V2. I can also trace back my initial problems to updates I have made in the last few days.

I have not tried to re-flash my SD card with a fresh image yet, though I suspect that would solve my problem as well.

Currently I am attempting to determine which updates break the LED matrix API so that I can roll them back. Please let me know if you determine which updates affect the Sense Hat, and which versions are stable to use currently.

XECDesign, 2023-03-22:
That would've been the kernel update. Something broke in the move to 6.1.

tx73github, 2023-03-24:
Same here, seems only to affect the LED matrix. 
Confirmed not working with 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux.

DMHas, 2023-03-24:
Thanks all for confirming. I open a new issue here: https://github.com/raspberrypi/firmware/issues/1798

pelwell, 2023-03-24:
This has been fixed today. You can install a beta kernel with `sudo rpi-update`.

davidbradway, 2023-03-26:
> This has been fixed today. You can install a beta kernel with `sudo rpi-update`.

I just installed the beta kernel. It fixed the LED matrix but seems to have broken the temperature and humidity measurements. The pressure sensor is still reading something, while the other two are "0". Can anyone reproduce my experience?

pelwell, 2023-03-27:
It has also been pointed out that the mmap support got broken back in 5.18, but https://github.com/raspberrypi/linux/pull/5404 should fix that.

DMHas, 2023-03-27:
> @pelwell , @DMHas : I just installed the beta kernel. It fixed the LED matrix but seems to have broken the temperature and humidity measurements. The pressure sensor is still reading something, while the other two are "0". Can anyone reproduce my experience?

I can't confirm the zero values after firmware update.

But my both RasPi's (2B & 4 with 8GB) and both SenseHat's (V1 & V2) shows this behavior sometimes. Turn off and turn on fix it (restart change nothing). The RasPi 2 shows the zero values often as the RasPi 4. It happens more, when 2 program's have access to temperature, air pressure and humidity at the same time. (May be it is depending from the CPU load or GPIO load.)

DMHas, 2023-03-27:
> The only issues that have been addressed are the cancelling of deferred I/O on close and the mmap() breakage. Please open a new issue if you are seeing other problems.

I will observe the behavior. If it happens again, I will open a new thread