astro-pi/python-sense-hat #140: Mismatched versions across GitHub, PyPI, docs site(s)

bbrewington, 2023-06-19:
Seeing some version discrepancies across sites, so figured this table view would be the best way to collapse it all.  I'm assuming everything needs to match the latest version in GitHub releases

Also, I'm new here and don't see a CI .yml file or maintenance instructions in a Wiki - so pardon my ignorance with this question: Would it make sense to set up automation so that when there is a new release in GH repo, it publishes updates to PyPI & readthedocs?  And if that was automated, it might help ensure everything matches for future releases

Page | sense-hat Version | Issues
---|---|---
GitHub Repo astro-pi/python-sense-hat | v2.5.1 (2022-11-21) | 1. Releases has version 2.5.1 ([link](https://github.com/astro-pi/python-sense-hat/releases))<br>2. __init__.py has version 2.4.0 ([link](https://github.com/astro-pi/python-sense-hat/blob/763813398258445363574ea47cab912ab43e89f2/sense_hat/__init__.py#L16))<br>3. setup.py has version 2.4.0 ([permalink](https://github.com/astro-pi/python-sense-hat/blob/763813398258445363574ea47cab912ab43e89f2/setup.py#L10))
PyPI ([link](https://pypi.org/project/sense-hat/)) | v2.4.0 (2022-08-09) | 1. is behind GitHub release
sense-hat.readthedocs.io ([link](https://sense-hat.readthedocs.io)) | v2.4.0 (2022-11-21) | 1. is behind GitHub release
pythonhosted.org/sense-hat ([link](https://pythonhosted.org/sense-hat/)) | v2.2.0 | 1. Now that readthedocs is live, this needs to be removed from this repo ([link](https://github.com/search?q=repo%3Aastro-pi%2Fpython-sense-hat%20pythonhosted&type=code))<br>2. Needs to be updated in [github.com/raspberrypi --> sense-hat](https://github.com/raspberrypi/documentation/blob/df9e2e889a68ae75ec7585cf9205e249c6df03fe/documentation/asciidoc/accessories/sense-hat/software.adoc?plain=1#L43)

bsimmo, 2023-06-19:
Have a look a #139, it says the others get bumped when RPF ask for it.

This may well be because AstroPi[AstroPi](https://astro-pi.org/)  use this and any major changes might alter the project at the time of year it is running.(ignoring any RPF Project documents that may also need a tweak)

SenseEmu may need a tweak too https://github.com/astro-pi/python-sense-emu
(and Trinket?)

bbrewington, 2023-06-20:
Got it. From here, I see two options to make sure that people in the future understand this... which one is preferable?

1. Document that versions may be different - in the repo, PyPI, and readthedocs
1. Set up automation so release is automated (and "release" means "go update all versions)

Or am I misunderstanding the issue linked above?

G3zz, 2023-10-11:
Thank you for making the table. I have aligned the version numbers everywhere (and released a new version for the new Astro Pi year), and requested a new release be prepared by RP OS.

G3zz, 2023-10-24:
The version numbers are now all aligned, so I will close this issue. Thanks!

closed 2023-10-24 by G3zz