astro-pi/python-sense-hat #143: Colour sensor not detected on raspberry Pi 5

ZGMFX20AR, 2024-02-23:
Hello, I'm experiencing an issue with the Sense HAT v2 on the Raspberry Pi 5.

When executing the i2cdetect -y 1 command on the Pi 5, it successfully detects all sensors except for the color sensor.

Conversely, on the Pi 4, the command detects all sensors including the color sensor, which has an I2C address of 0x29.

XECDesign, 2024-04-15:
Are you using the same Sense HAT in both cases? I think newer Sense HATs have the colour sensor on 0x39. Tested on a pi5 here and the output was as expected.

yupm, 2024-05-02:
I think the library is not updated in this case.

```
from sense_hat import SenseHat
from time import sleep

sense = SenseHat()
sense.color.gain = 60
sense.color.integration_cycles = 64

while True:
    sleep(2 * sense.colour.integration_time)
    red, green, blue, clear = sense.colour.colour # readings scaled to 0-256
    print(f"R: {red}, G: {green}, B: {blue}, C: {clear}")
```
Error:
```
Traceback (most recent call last):
  File "/home/kisuidev/csense.py", line 5, in <module>
    sense.color.gain = 60
    ^^^^^^^^^^^
AttributeError: 'SenseHat' object has no attribute 'color'
```
Tested on Ubuntu 24.04

bsimmo, 2024-05-03:
Spell it correctly and it should work?

colour

You have it correct later.

yupm, 2024-05-05:
Erm... the official documentation is using color though 
https://projects.raspberrypi.org/en/projects/getting-started-with-the-sense-hat/7
[image]

Anyway I managed to fix it by just using Rasbian as the default OS. 
After setting up Ubuntu via docker and fooling around with building this repo and the RTIMU repo from source, I am thinking the Ubuntu 24.04 issue might be a result of outdated packages in the repo..

bsimmo, 2024-05-06:
They must have changed it at some point, it never used to work. At least I
don't think it did.

Anyway, glad it working.

For the record, it's not Raspbian, it is RaspberryPi OS. It would be better
to call it Debian as it's not using Raspbian any more. (see history in some
old blog, also Raspbian was 32bit only).

bsimmo, 2024-05-06:
It's seems it was 4 years ago they added color as spelling :-)