astro-pi/python-sense-hat #144: Sense Hat support for Ubuntu (e.g. version 23.10 mantic)

aotearoa-david, 2024-07-26:
```
davidg@pinas:~/pinas/Projects/301_Sense_HAT_Python$ sudo python3 ./hello.py 
[sudo] password for davidg: 
Hello, world!
Traceback (most recent call last):
  File "/home/davidg/pinas/Projects/301_Sense_HAT_Python/./hello.py", line 5, in <module>
    sense = SenseHat()
            ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/sense_hat/sense_hat.py", line 39, in __init__
    raise OSError('Cannot detect %s device' % self.SENSE_HAT_FB_NAME)
OSError: Cannot detect RPi-Sense FB device
davidg@pinas:~/pinas/Projects/301_Sense_HAT_Python$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 23.10
Release:	23.10
Codename:	mantic
```

Following along with tutorial [https://projects.raspberrypi.org/en/projects/getting-started-with-the-sense-hat](https://projects.raspberrypi.org/en/projects/getting-started-with-the-sense-hat)

The code I was using is not very fancy; just:

**hello.py**

```
from sense_hat import SenseHat

# print hello world
print("Hello, world!")
sense = SenseHat()

sense.show_message("Let's have piklets for breakfast!")
```

I'm aware that I should be able to switch to [https://www.raspbian.org/](https://www.raspbian.org/) but I would like to find others to help me put time/effort into getting this to work more broadly, hence logging this issue.

bsimmo, 2024-07-27:
Ask on the Pi forum. This isn't really a bug. It's a setup with Ubuntu as
you'll need to install it, or activate the overlay, and use Python however
Ubuntu wants you to use it.
There would normally be no need to run eleveted either. At least not in
RPiOS.

I suggest you grab another sd card, us RPiOS and see how it works there,
then work backwards once you have the feel.

aotearoa-david, 2024-07-27:
Hi @bsimmo - Amazed at the quick reply. Yes, if you want to close this ticket as it is not a bug that is fine. I have this on my backlog of hobby tasks to do.

Hi subject matter experts, if you want to pair up on this one please get in touch.

closed 2024-07-27 by aotearoa-david