astro-pi/python-sense-hat #148: SenseHat object cannot be instantiated more than once.

nralbers, 2025-03-18:
If more than a single SenseHat object is instantiated on a machine any calls to sensor data will return 0 instead of the actual sensor value.

Simple test:

```python
from sense_hat import SenseHat
import unittest

class TestSingleton(unittest.TestCase):
    def test_singleton(self):
        sense1 = SenseHat()
        sense2 = SenseHat()
        self.assertEqual(sense1.temperature, sense2.temperature)    

if __name__ == "__main__":
    unittest.main()
```

This will fail on the current codebase.
I've added pull request #147 to at least prevent this occuring in a single python program by converting the SenseHat object to a thread-safe Singleton, but this will not prevent the issue arriving if multiple projects are using the SenseHat object on the same machine.

bsimmo, 2025-03-18:
When would you have two sensehats, they would share the same i2c and other
port/device numbers etc.

nralbers, 2025-03-18:
> When would you have two sensehats, they would share the same i2c and other
> port/device numbers etc.
> […](#)

This is not for having two physical sense hats, but for the case where the SenseHat object is instantiated in multiple places, for example  if the sense object is being used in multi threaded code like in a web api running multiple worker threads.
The changes I've tested here are a guard rail because the object cannot handle being instantiated twice.